StanfordVLSI/dragonphy2 #4: mux_gf does not appear to block all glitches

sgherbst, 2020-04-28:
For example, when the input select code is changed from "0" to "2", a glitch with two positive edges is seen.  This is particularly critical because it can cause sampler sequence to slip in an unpredictable way, breaking the ordering of the retimer.

sjkim85, 2020-04-28:
I think it makes sense to me. If the input select code of a mux4_gf is changed from 0 to 2, that means (4*16) steps of PI input code was changed. Which is huge amount of change and I think it will not be happened in actual CDR tracking loop.

sjkim85, 2020-04-28:
By the way, did you add a new isolated test for the mux4_gf only to see the glitch behavior of the cell?
In current version of test benches that I have (new_tests/GLITCH) this behavior is not detected because the situation not happens.

sgherbst, 2020-04-28:
Sure, I'll add an isolated test for mux4_gf and will look at a change from "0" to "1" to see if the problem persists.

The system-level test GLITCH only looks for glitches in steady-state (this was an issue in the past).  Part of the reason that I am filing this issue is that I want to update the GLITCH test to check for glitches at any time.  However I can't do this yet because there are glitches on code transitions.

As far as whether this is a realistic scenario, I'm seeing this problem during startup when PI codes are changed from a default value to a custom value.

sjkim85, 2020-04-28:
FYI, this was the order of register setting when I did the lab test.
global_reset_b =1 --> set_PI_code --> toggling en_v2t from 0 to 1
(basically, for all test, the en_v2t was toggled from 0 to 1 whenever any register setting is changed)

sgherbst, 2020-05-01:
I added a test to investigate the isolated behavior of ``mux_gf4``.  The test is called ``MUX4_GF_ONLY`` and is currently in the ``glitch_tests`` branch.  In this test, the mux input consists of 4 clocks, each delayed 30ps with respect to the previous clock (with a little randomization make sure errors don't slip through the cracks due to lucky arithmetic cancellations).  The ``sel`` code of the mux is swept up and down between min and max values, one code at a time.  While this is happening, the test monitors for glitches (defined as a deviation from nominal period / low / high width of 40ps or greater, since the clocks are spaced by about 30 ps).

After adding this test, the ``mux4_gf`` block does appear to be glitch free, so I am closing this issue.  However, at a system level (PI-level), there are still some glitches, as I'll explain in comments for issue #21

closed 2020-05-01 by sgherbst