StanfordVLSI/dragonphy2 #5: ctl_dcdl_late and ctl_dcdl_early default to zero -- is this expected?

sgherbst, 2020-04-28:
I noticed that the delay control signals for the V2T early and late clocks are both zero by default.  Is this expected?  In my simulations I'm seeing that clk_v2t, clk_v2t_early, and clk_v2t_late all happen at the same time.

sjkim85, 2020-04-28:
The default value of the ctl_dcdl_early/late should be zero but I think this problem is due to the absence of cpu model of the new cell "mux_fixed.sv" (so this mux has zero delay now). I will add cpu delay models for newly added cells for the next PR.

sgherbst, 2020-04-28:
Thanks for looking into it.  The branch where I saw this (``ac_test``) does not have the ``mux_fixed`` cell, but it does have non-zero delays for all gates that are instantiated.  However, it could be that in the real design, the time difference comes from gates that are inferred (i.e. synthesized) rather than explictly instantiated.  Interested to hear what you find.

sgherbst, 2020-05-01:
Looks like this is fixed after adding delay to ``mux_fixed``.  @sjkim85 would you mind confirming that only the negative edges are supposed to occur at different times?  If so then I will close this issue.
[image]

sjkim85, 2020-05-01:
Yes. It is intended. Thanks

sgherbst, 2020-05-01:
Great, thanks for confirming.  Closing the issue.

closed 2020-05-01 by sgherbst