StanfordVLSI/dragonphy2 #6: ENOB for TDC is low

sgherbst, 2020-04-28:
Simulated ENOB is currently ~4.85 for the AC test (Fstim=1.023GHz).  Could be a modeling, testing, or design issue, but it's important to log here because this is an important performance metric.

sjkim85, 2020-04-28:
I will look AC test results. I may be due to the obsolete setting of V2T.

sgherbst, 2020-04-28:
Thanks.  Two possible causes:
1.  I have attempted to align PI clocks, but the alignment may not be perfect.  If you have suggestions on the tuning method to get exact spacing of all 16 samples, I would be interested to hear them.  I think this is important for lab testing too.
2. TDC nominal delay, standard deviation, or jitter values may be out of date.

sjkim85, 2020-04-28:
1) The yaw that I did for the lab testing is the following 
Did the sine regression with data comes from each ADC slice (16 sinusoidal regression in total) and estimated phase of each regression. I tweaked the PI code until I get uniform phase spacing.

2) I think Td_inv_nom = 15p, std=1p, jit =0.1p is reasonable value. One more thing I did in the lab testing was I swept ctl_dcdl_coarse to get the better ENOB (delay step of the dcdl_coarse is roughly ~40ps)

sgherbst, 2020-10-07:
This was resolved awhile ago.

closed 2020-10-07 by sgherbst