StanfordVLSI/dragonphy2 #21: Review GLITCH test

sgherbst, 2020-04-30:
The expectation is that changes of +/- 1 PI code should never cause a glitch in the PI output.  I think the current GLITCH test only checks for glitches in steady-state, so this needs to be updated.

sgherbst, 2020-05-01:
I updated the test to do this, and found some glitches, so I'm re-assigning to @sjkim85.  To reproduce the issue, check out the ``glitch_tests`` branch and run ``pytest tests/new_tests/GLITCH -s``.  Two different kinds of errors are seen:
1. True glitches (i.e., very short pulses and multiple clock edges).  These only occur at a few specific code transitions, like 496 -> 495 and 240 -> 239.
2. Low-resolution issues, where certain code transitions cause a much bigger change in phase than expected (~10ps instead of ~1ps).  An example would be the 255 -> 256 transition.  This kind of error occurs much more frequently on negative code transitions than positive code transitions.  Also, it appears to have a periodic behavior while codes are decreasing (i.e., every "N" codes there is a big step)

sjkim85, 2020-05-01:
I was trying to see waveforms but I got this errors.
(I ran the pytest  in the glitch_tests branch as you suggested after I did "python make.py --view fpga)
[cid:35241fae-bd5f-43ea-b73d-0b9ec09f645b]

sgherbst, 2020-05-01:
You can send me a Slack message or email with the error message and we can debug from there.  In general, I suggest re-running ``pip install -e .`` in the top-level directory to reinstall DragonPHY when you see errors, since there may be new package dependencies that were added.  Also, you'll need to switch to the ``pwl_cos`` branch on DaVE since there is an open pull request.  When in doubt, the definitive install steps can always be found in the ``regress.sh`` script.

sgherbst, 2020-05-02:
Note: while @sjkim85 is debugging this, I marked the GLITCH test "work in progress".  To run the GLITCH test, add ``--runwip`` to the normal arguments you pass to ``pytest`` (otherwise the test will be skipped).  Then, as a condition for resolving this issue, we need to remove the "work in progress" marker from the test.

sgherbst, 2020-05-07:
@sjkim85 Thanks for all of your work on this issue.  I noticed that it's still marked as "work-in-progress", meaning that the GLITCH test is not being run in the regression.  Could you update this in the next PR?
This can be done by changing ``pytest.mark.wip`` in ``test_glitch.py`` to:
```python
@pytest.mark.parametrize((), [pytest.param(marks=pytest.mark.slow) if SIMULATOR=='vivado' else ()])
```
Should just be a one line change but it could be more involved if it turns out the that GLITCH test is not passing on the regression server.

sjkim85, 2020-05-07:
Sure.
I will push this change with dragonphy_top (add mdll) soon.

Thanks

-Sung-Jin

sgherbst, 2020-05-14:
Closing since this was resolved a few days ago in the PR mentioned about dragonphy_top/mdll

closed 2020-05-14 by sgherbst