StanfordVLSI/dragonphy2 #60: Where does JTAG fit into the reset sequence?

sgherbst, 2020-05-01:
During a meeting a few days ago, we defined the reset sequence as:
1. De-assert ``ext_rstb``
2. De-assert ``int_rstb`` (over JTAG)
3. Assert ``en_inbuf``
4. Assert ``en_gf``
5. Assert ``en_v2t``
But I realized this doesn't include the JTAG reset pin, ``phy_trst_n``.  Where does that belong in this sequence?

zamyers, 2020-05-01:
the JTAG reset pin is controlled by the JTAG controller.

-Zach

sgherbst, 2020-05-01:
Sorry, to clarify I mean *when* this should happen, rather than *how*.  It definitely needs to happen before step 2 (since that involves JTAG communication), so I think my question can be simplified to "which should be de-asserted first, ``ext_rstb`` or ``phy_trst_n``"

zamyers, 2020-05-01:
It happens after ext_rstb is de-asserted. I can double check the logic but I believe the ext_rstb acts as a standard reset inside the JTAG and the JTAG's trst_n's primary function is to reset JTAG state machine into a known state.

-Zach

sgherbst, 2020-05-01:
Great, thanks.  In that case I'll update the reset sequence to
1. De-assert ``ext_rstb``
2. De-assert ``phy_trst_n``
3. De-assert ``int_rstb`` (over JTAG)
4. Assert ``en_inbuf``
5. Assert ``en_gf``
6. Assert ``en_v2t``

closed 2020-05-01 by sgherbst