StanfordVLSI/dragonphy2 #61: synopsys translate_off

sgherbst, 2020-05-01:
Just noticed that this is used in at least two places to initialize values.  I think we need to be really careful to make sure that we're not relying on these initial values in the real design.  Here are the places where I found this:

https://github.com/StanfordVLSI/dragonphy2/blob/2d280b99c7884aedabea331e1e2498c6bb49c457/vlog/new_chip_src/analog_core/V2T_clock_gen.sv#L45-L49

https://github.com/StanfordVLSI/dragonphy2/blob/2d280b99c7884aedabea331e1e2498c6bb49c457/vlog/new_chip_src/analog_core/PI_local_encoder.sv#L36-L42

If possible, I suggest we remove these non-synthesizable initializations.

bclim, 2020-05-01:
Sometimes it is necessary if the counter reg doesn’t have reset. But it looks like those regs have an reset input. So, as Steven suggested, it would be good to get rid of the initial statement and reset the counter properly.
Another way (less conservative way) in general is to randomly initialize it using $urandom().

btw, I looked at the code, but I don’t quite understand the code to have two async resets, and I was wondering how that is mapped to a gate.

bclim, 2020-05-01:
That will be mapped to a flop with both set and reset. But some stdcell providers might not have that kind of flops

sgherbst, 2020-05-08:
Done

closed 2020-05-08 by sgherbst