StanfordVLSI/dragonphy2 #64: Set PI control codes in tests through normal JTAG debug signals

sgherbst, 2020-05-02:
Right now PI-related tests set the PI code through a signal that is not part of the JTAG debug interface.  Once issue #62 is resolved they can be set directly through normal JTAG debug signals, which is a better representation of what we would actually be able to do in the lab.

zamyers, 2020-05-07:
This exists the test_dev branch and will be closed upon PR of that branch.

sgherbst, 2020-10-07:
This still doesn't appear to be resolved in the following tests:
1. ``cpu_system_tests/pi``
2. ``cpu_system_tests/pi_pm``

The glitch test also has an unusual way of setting the PI control code, but I think that's what we want in that case, since it makes it easier to create different timing scenarios.