StanfordVLSI/dragonphy2 #119: Sparam scale is very large

zamyers, 2020-09-30:
When you use the s4p_to_impulse, the resulting output is very large (proportional) ~ with cursor peaks of 1e9.

I assume this falls out of some inverse FFT somewhere not being normalized by the number of bins?

sgherbst, 2020-09-30:
Great comment -- this gets at the difference between an impulse response and a pulse response.  The impulse response will typically have a large magnitude, since it is the derivative of the step response, which is very fast.  I think that you mostly likely want to use the pulse response instead, which is the channel response to a single "1" bit spanning a unit interval.

I added some code to the `s4p` branch that makes it straightforward to do this, using our existing `Channel` class:
```python
chan = Channel(channel_type='s4p', sampl_rate=10e12, resp_depth=200000,
    s4p=file_name, zs=50, zl=50)
_, pulse = chan.get_pulse_resp(f_sig=16e9, resp_depth=250, t_delay=0)
```

zamyers, 2020-09-30:
Pulse Response is a suitable solution.

closed 2020-09-30 by zamyers