StanfordVLSI/dragonphy2 #137: Add output buffer (and termination?) for TX

sgherbst, 2020-10-24:
Just wanted to log this issue here so that we can keep track of it.  I'm assigning @CansWang since he is the TX designer, and it would be great to hear thoughts from @sjkim85 as well because he designed the differential output buffer used in the digital core.

sjkim85, 2020-10-24:
The output buffer in DCORE is just inverter chain for monitoring purpose and its output impedance is not matched at all.
I thought our TX contains output driver stage to drive channel. Am I right Can?

-Sung-Jin

Sent via the Samsung Galaxy S8+, an AT&T 5G Evolution capable smartphone
Android용 Outlook<https://aka.ms/ghei36> 다운로드

CansWang, 2020-10-24:
> The output buffer in DCORE is just inverter chain for monitoring purpose and its output impedance is not matched at all. I thought our TX contains output driver stage to drive channel. Am I right [...]

The Tx output stage needs to be designed together with the termination and integrate into `tx_top.sv`.
There are essentially two ways of designing it (?) 

since we want to minimize the delay and number of stages, also make the on resistance + termination to be closer to 50 ohm (single ended)
1) design this in virtuoso
2) synthesize it, (I am not familiar with this method), but we should be able to specify the fanout (parasitic at the output) and unit size (determined by sheet resistance of diffusion)

Correct me if anything doesn't make sense to you, thanks!

sgherbst, 2020-11-01:
Closing as this was resolved in #146

closed 2020-11-01 by sgherbst