StanfordVLSI/dragonphy2 #140: Remove combo logic loop in qr_4t1_mux_top

sgherbst, 2020-10-27:
There is currently a combo logic loop in ``qr_4t1_mux_top``, where ``data`` appears on both the ``RHS`` and ``LHS`` of an ``assign`` statement:
https://github.com/StanfordVLSI/dragonphy2/blob/1f1c97ba5f52445fe06aa39ac60585e053916806/vlog/chip_src/tx_16t1/qr_4t1_mux_top.sv#L47

I believe this was added to prevent glitches if clocks are non-overlapping, but there may be other ways to deal with this, for example given the 4-bit select signal ``{Q, I, QB, IB}``:
* ``4'b1100`` or ``4'b0100``: choose ``D1MQ``
* ``4'b0110`` or ``4'b0010``: choose ``D1MI``
* ``4'b0011`` or ``4'b0001``: choose ``D2MQB``
* ``4'b1001`` or ``4'b1000``: choose ``D2MIB``

However, because this is such a critical block, it would be helpful to think about what the intended digital circuit is, so that we can make sure that is what will be produced by the Verilog.

sgherbst, 2020-10-27:
Feedback from Mark was to implement as the OR of 3-input ANDs, i.e.
```verilog
assign data = ((clk_Q  & clk_I  & D1MQ ) |
               (clk_I  & clk_QB & D1MI ) |
               (clk_QB & clk_IB & D2MQB) |
               (clk_IB & clk_Q  & D2MIB)); 
```
This removes the combo loop issue and shows the synthesis intent more explicitly.  There is still a potential for glitches, but he suggested we evaluate its impact to see whether we need to do anything specific to address it.

CansWang, 2020-10-27:
> Feedback from Mark was to implement as the OR of 3-input ANDs, i.e.
>
> ```verilog
> assign data = ((clk_Q  & clk_I  & D1MQ ) |
>                (clk_I  & clk_QB & D1MI ) |
>                (clk_QB & clk_IB [...]

The glitches are expected to happen, however, it is normally 1-2ps since otherwise we would adjust the PI control code to minimize it. A pulse with 1-2 ps width would be negligible as it is too sharp to propagate through parasitics.

Thanks for checking with Mark.

closed 2020-10-28 by sgherbst