StanfordVLSI/dragonphy2 #143: two additional primary IOs for DCORE

sjkim85, 2020-10-29:
- two primary IOs (external control) should be added  for DCORE's new function (freq_ramp, lvl_cross)
- determine whether these pins are in dcore_intf or dragonphy_top directly

sgherbst, 2020-10-29:
@sjkim85 these pins appear in the top-level I/O of ``dragonphy_top``:
https://github.com/StanfordVLSI/dragonphy2/blob/724e5662aeba40a329fa2fca84772e03b721a684/vlog/chip_src/top/dragonphy_top.sv#L37-L39

and they are wired directly into ``idcore`` (i.e., not through ``dcore_intf``)
https://github.com/StanfordVLSI/dragonphy2/blob/724e5662aeba40a329fa2fca84772e03b721a684/vlog/chip_src/top/dragonphy_top.sv#L221-L222

does that address your request, or were you asking about something else?

sjkim85, 2020-10-29:
@sgherbst, It is my mistake. Thank you for your clarification.

closed 2020-10-29 by sjkim85